lnis-uofu/OpenFPGA #1502: vpr is so slow with large circuit

ducminhnguyen123, 2024-01-03:
I am implementing a large circuit on OpenFPGA and when in VPR step, it tooks a lot of time (particularly 222590 seconds) to run. In vtr documents, I see that there is a option to optimize my circuit but it is still experimental (--router_algorithm parallel). I want to ask that how long is this option completed? Or could you give me some advices for me to improve my circuits?

Thanks,

ducminhnguyen123, 2024-01-03:
There are some options which you wrote in your documents but it is still not good enough. I have read your c++ code but it is difficult to change your routing algorithm. Please, can you give me some advice or some hints for this issue?

ducminhnguyen123, 2024-01-04:
please comment if you have any concern
@tangxifan

tangxifan, 2024-01-12:
@ducminhnguyen123 

VPR has been tested on many benchmarks where number of LUTs is up to 1M. See details in the VTR8 research paper.

https://www.eecg.utoronto.ca/~kmurray/vtr/vtr8_trets.pdf

You may find details in the Table 28 (See a screenshot below). It can accomplish in 1 day even for the largest. Most of them can be finished in 2 hours. The QoR is continuously monitored on VTR master branch.

[image]

So, if you are experiencing such issue, it mostly is due to some wrong settings. My suggestion is to share your log file or architecture XML. It helps to find out why.

ducminhnguyen123, 2024-01-15:
thanks a lot with your answer, can you tell me how many luts per clb you config, pls?

lehaifeng000, 2024-06-26:
@ducminhnguyen123 try nextpnr