lnis-uofu/OpenFPGA #1503: ql_memory_bank reading bitstream back

mustafaarslan0, 2024-01-03:
I use ql_memory_bank shift_register configuration protocol with SRAM cells. SRAM cells have BL, WL and WLR pins as described in [image].

We need read enable signal for reading SRAM but I think WLR is overkill for this function. Wouldn't it be better if we use 1-bit wide read_enable signal (connected to all SRAMs) and WL together. WL is for decoding and read_enable signal is for deciding write or read operation.

Also BL should be defined as inout for bidirectional read/write operations.

Thank you,
Best regards.

tangxifan, 2024-01-12:
@mustafaarslan0 

In your case, you may consider to create a global signal ``EN`` for your SRAM.
Take the example (focus on the CFG_EN port)

https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_model_examples/#configuration-chain-flip-flop-with-configure-enable-signals

mustafaarslan0, 2024-04-03:
@tangxifan creating a global EN signal works. But BL should be defined as inout for bidirectional read/write operations.

mustafaarslan0, 2024-12-04:
[data read signal in OpenFPGA sram cell library](https://github.com/lnis-uofu/OpenFPGA/blob/68433d3a38d8a320bf1136bbdc78bb634642e82e/openfpga_flow/openfpga_cell_library/verilog/sram.v#L272) does not drive any nets. How to read data via SRAM_RE cell and qlbank configuration protocol? 

OpenFPGA does not generate another BL for read (it expects only one BL). I considered adding a BLB line and reading data from it, but OpenFPGA leaves the BLB lines undriven.

@tangxifan