lnis-uofu/OpenFPGA #1507: Question about BRAM preload data content

chungshien, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
Hi Xifan,

I would like to understand how BRAM preload data content is being generated. Hope you can help me to understand that

For example: 

1. In the project RTL, we will initialize the BRAM with random content from file mem1.mem

[image]

2. In the systhesis eblif file, I could see that the content is initialized as part of "param INIT_i"

[image]

Will this BRAM Preload data content being generated as part of bitstream? Do we have BRAM bitstream like what happened in fabric (for example fabric will generate fabric_bitstream.bit)? 

Thanks

chungshien, 2024-01-08:
@alaindargelas

alaindargelas, 2024-01-09:
@tangxifan we can see the memory content being preserved by VPR and make it into the post-route netlist, but we cannot find it in the bitstream.

tangxifan, 2024-01-12:
@alaindargelas @chungshien 

There are multiple ways to implement BRAM preloading.
In your way, you may refer to the bitstream setting file:

https://openfpga.readthedocs.io/en/master/manual/file_formats/bitstream_setting/

where you can extract information from eBLIF file and use it in your bitstream.

chungshien, 2024-03-10:
Closing this issue as https://github.com/lnis-uofu/OpenFPGA/pull/1566 had been merged.

closed 2024-03-10 by chungshien