lnis-uofu/OpenFPGA #1513: different area between 10LUTs and 1LUT

ducminhnguyen123, 2024-01-12:
when I use 1 LUT or many LUTs in 1 clb, how is the area affected and which value refers to that area. (area of 10LUTs and area of 1 LUTs).

ducminhnguyen123, 2024-01-12:
@tangxifan I will be grateful if you give me the answers. 

best regard

tangxifan, 2024-01-12:
@ducminhnguyen123 

The area of 10 LUTs is simply 10x bigger than 1 LUT. If you count in local routing, you may see more area overhead.
However, in modern FPGAs, no one is using 1 LUT per CLB. It is not efficient as it demands a lot of global routing to accomplish your implementation. A good number of LUTs per CLB ranges from 6 to 12,  depending on your needs.

ducminhnguyen123, 2024-01-15:
thanks a lot about your answer. Can you give me answers for my another question. When running the same RTL with auto size, how will the chip area be affected when the luts are different, specifically what trade-offs must be made between the clb blocks, switch blocks, connection blocks in terms of area and how do the internal components affect area? In addition, I want more detail informations about the value which refer to the area I have talked above? Can I see it in openfpgashell.log?

tangxifan, 2024-01-15:
@ducminhnguyen123 

It will change a lot when these parameters are altered. To understand the trade-offs, I suggest you to give a read on the principles of FPGA architecture: https://tspace.library.utoronto.ca/handle/1807/12758

You can find the details in the openfpgashell.log in your runtime directory.

ducminhnguyen123, 2024-01-19:
@tangxifan thanks for your answers. I have an another question. I want to know the practical area value of FPGA by mm2 unit. Does the area here relate to the number of LUTs we use (the area of 10 LUTs different to the area of 1 LUTs ?). Where can I see this actual area value? ( Can you give me the LOG file which has the value)?