lnis-uofu/OpenFPGA #1528: Full testbench is not working

ducminhnguyen123, 2024-01-22:
Dear tangxifan, 

I have used this folder: openfpga_flow/tasks/basic_tests/k4_series/k4n4_fracff to run openfpga. Preconfigured testbench runs and give expected results. However, within same task I also generate full testbench but it froze at vvp_verification. When I simulated this full testbench on questasim tool, it is broken like this picture below.
[image]

What would be my problem? How can I solve this issue? 

I also comment 2 line (line 56, line 57) like this picture below but it can not still work
[image]

Best regard,

Duc Tiger

chaitalisathe, 2024-01-29:
Hello! I have had similar experience in openfpga_flow/tasks/basic_tests/full_testbench/configuration_chain/.
Testbench is freezing during vvp_verification.
In my scenario bitstream is of size 12788 and after 4570 bits, bitstream is not loaded into ccff_head and simulation freezes.

Have you found the issue? 

Please let me know how to resolve this.

Thank you,

Chaitali

Lukemagik, 2024-03-10:
Hello,

I am encountering a similar issue with the  openfpga_flow/fpga_verilog/dsp/single_mode_mult_8x8 task. Specifically, the full testbench stalls at a certain point when loading the bitstream, while the formal testbench operates as expected. Interestingly, this stalling also occurs with a full_tb on an architecture that I manually modified.

Have you been able to identify the root cause of this problem?

chaitalisathe, 2024-04-05:
@Lukemagik any luck with this issue? I haven't found the solution yet.

Lukemagik, 2024-04-05:
@chaitalisathe 
Hi, I think the issue is related to the scan-chain configuration protocol. By using a different protocol, in our case the one referred to as frame-based, we have resolved it.

chaitalisathe, 2024-04-09:
@Lukemagik Thank you! I tried configuration frame and it worked. I was able to generate fabric, bitstream and load into FPGA. and now vvp verification is failing. It's not freezing but I am not getting any output signals from the fabric. 
VVP verification is failing for all the designs with configuration frame protocol which were working perfectly before on configuration chain.
Did you face a similar issue?

chaitalisathe, 2024-04-09:
[image]

Lukemagik, 2024-04-09:
Hi, @chaitalisathe , which benchmarks are not working?
Did you modify the .xml file that previously used the scan-chain protocol, or did you use directly an architecture where it was already implemented, like the k4_N4_40nm_frame_openfpga.xml?

chaitalisathe, 2024-04-09:
@Lukemagik I am running this script  - 
python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/full_testbench/configuration_frame

task.config file under configuration_frame folder reads as follows-

# = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =
# timeout_each_job : FPGA Task script splits fpga flow into multiple jobs
# Each job execute fpga_flow script on combination of architecture & benchmark
# timeout_each_job is timeout for each job
# = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = = =

[GENERAL]
run_engine=openfpga_shell
power_tech_file = ${PATH:OPENFPGA_PATH}/openfpga_flow/tech/PTM_45nm/45nm.xml
power_analysis = true
spice_output=false
verilog_output=true
timeout_each_job = 20*60
fpga_flow=yosys_vpr

[OpenFPGA_SHELL]
openfpga_shell_template=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_shell_scripts/write_full_testbench_example_script.openfpga
openfpga_arch_file=**${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_arch/k4_N4_40nm_frame_openfpga.xml**
openfpga_sim_setting_file=${PATH:OPENFPGA_PATH}/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml
openfpga_vpr_device_layout=
openfpga_fast_configuration=

[ARCHITECTURES]
arch0=**${PATH:OPENFPGA_PATH}/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml**

[BENCHMARKS]
bench5=${PATH:OPENFPGA_PATH}/openfpga_flow/benchmarks**/micro_benchmark/adder/adder_16/adder_16.v**

[SYNTHESIS_PARAM]
bench_read_verilog_options_common = -nolatches
bench5_top = adder_16
bench5_chan_width = 300
[SCRIPT_PARAM_MIN_ROUTE_CHAN_WIDTH]
end_flow_with_test=