lnis-uofu/OpenFPGA #1697: 'CHANX == rr_graph.node_type(src_node) || CHANY == rr_graph.node_type(src_node)' failed.

kpatinos-intel, 2024-06-07:
Hello, I am trying to define and learn how to build different architectures. However, I am facing an Error I don't know how to solve or manage. 

/home/kpatinos/Intel/OpenFPGA/vtr-verilog-to-routing/vpr/src/tileable_rr_graph/rr_gsb.cpp:894 sort_ipin_node_in_edges: Assertion 'CHANX == rr_graph.node_type(src_node) || CHANY == rr_graph.node_type(src_node)' failed.

This is my flow configuration:

```
# Run VPR for the 'and' design
pcf2place --pcf ${OPENFPGA_PCF_PATH}${TOP_MODULE}.pcf --blif ${VPR_TESTBENCH_BLIF} --pin_table ${OPENFPGA_PIN_TABLE} --fpga_io_map ${OPENFPGA_IO_MAP_FILE} --fpga_fix_pins ${TOP_MODULE}_fixed.place --pin_table_direction_convention ${OPENFPGA_PIN_TABLE_DIRECTION_CONVENTION} --verbose

#--write_rr_graph example_rr_graph.xml
vpr ${VPR_ARCH_FILE} ${VPR_TESTBENCH_BLIF} --clock_modeling route \
                                           --device ${OPENFPGA_VPR_DEVICE_LAYOUT} \
                                           --sdc_file ${OPENFPGA_DESIGN_SDC_FILE} \
                                           --fix_clusters ${TOP_MODULE}_fixed.place \
                                           --gen_post_synthesis_netlist on \
                                           --timing_report_detail detailed

# Read OpenFPGA architecture definition
read_openfpga_arch -f ${OPENFPGA_ARCH_FILE}

# Read OpenFPGA simulation settings
read_openfpga_simulation_setting -f ${OPENFPGA_SIM_SETTING_FILE}

# Annotate the OpenFPGA architecture to VPR data base
# to debug use --verbose options
link_openfpga_arch --sort_gsb_chan_node_in_edges --verbose

# Check and correct any naming conflicts in the BLIF netlist
check_netlist_naming_conflict --fix --report ./netlist_renaming.xml

``` 
It could be really useful any insight what I should check on my VPR-ARCH or the OpenFPGA arch files.

Thank you!

tangxifan, 2024-06-12:
@kpatinos-intel If you are using the latest master, the restriction should already be removed.

kpatinos-intel, 2024-06-12:
Thank you! No I was using an old commit, so now with the latest master that error is not showing, but now I am facing:  /home/kpatinos/Intel/OpenFPGA/openfpga/src/mux_lib/mux_library_builder.cpp:52 build_routing_arch_mux_library: Assertion '1 == driver_switches.size()' failed.

closed 2024-06-12 by kpatinos-intel