lnis-uofu/OpenFPGA #1723: Bitstream updates and reconfiguration support in OpenFPGA

ohault, 2024-06-23:
FPGA bitstream generation and loading are becoming quite mature topics, but what about the issue of updates? 

To dynamically modify a partial circuit on the reconfigurable structure at runtime, it would require the implementation of some dedicated functionality in the FPGAs.

How could such facilities be supported by OpenFPGA currently or in the future?

tangxifan, 2024-06-23:
@ohault It is a good point, and we are aiming to it. However, in my view, it requires both hardware and software supports. Some of them have already been done but still you may need more.

## Hardware

We have upgraded our configuration protocols to support partial reconfiguration. You can reconfigure part of an FPGA fabric.

- CCFF: Now each configuration chain can be controlled by a dedicated programmable block.  See details at: https://openfpga.readthedocs.io/en/master/manual/arch_lang/config_protocol/#configuration-chain-example
- Memory bank: Each row of configuration bits can be loaded by controlled a dedicated word line. See details: https://openfpga.readthedocs.io/en/master/manual/arch_lang/config_protocol/#quicklogic-memory-bank-example

Something is still missing could be gating the signals when loading a different bitstream. 

## Software

- Placement may be reworked so that partial reconfiguration is only required in a small region of your FPGA.
- Compare and merge bitstreams from two or more applications, with which it is easy to reconfigure an FPGA.

Here are my thoughts on this topic. 
If you see anything missing, feel free to contact me.

sumimehra, 2025-06-02:
@tangxifan Thanks for the detailed update. As I understand, the configuration protocol in OpenFPGA supports partial reconfiguration. So, is it currently possible to perform partial reconfiguration in the latest OpenFPGA releases, including the generation of bitstreams and runtime support? If yes, are there any example flows or documentation available?