lnis-uofu/OpenFPGA #1738: Does OpenFPGA support global short-wired outputs?

yunuseryilmaz18, 2024-07-03:
> **Describe the bug**
> A clear and concise description of what the bug is.

Hello,
While I was reading the docs, I encountered with [FPGA I/O port](https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#fpga-i-o-port) section and I want to try with AXI slave block made by me. I've made an OpenFPGA task and run it but I got `Circuit port (type=output) of model (name=axi_slave) is defined as global output port but not an I/O!` errors for my global short-wired outputs. Are global short-wired outputs supported right now? What should I do to solve it?

> Which part of OpenFPGA is buggy
> [X] OpenFPGA flow

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: d6fe414cd177be9e8e91adc46cc2c33c91b20062
> 2. Execute OpenFPGA task in the files.
> 3. See error

> **Expected behavior**
> I want to add an AXI slave block for my FPGA. AXI signals will be input and output of the FPGA fabric aside from the standard GPIO circuits.

> **Screenshots**
> [image]

> **Enviornment (please complete the following information):**

>  - OS: RHEL 8 or Ubuntu 20.04 (I've tried them on both)

>  - Compiler: GCC 8

> - Version: d6fe414cd177be9e8e91adc46cc2c33c91b20062

> **Additional context**
> Task files are here: [fpga_axi.zip](https://github.com/user-attachments/files/16082262/fpga_axi.zip)

tangxifan, 2024-07-05:
@yunuseryilmaz18 It looks that you may miss the ``is_io=false`` attribute in your XML file. You may refer to the following example: https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/spypad/config/task.conf

yunuseryilmaz18, 2024-07-06:
> @yunuseryilmaz18 It looks that you may miss the `is_io=false` attribute in your XML file. You may refer to the following example: https://github.com/lnis-uofu/OpenFPGA/blob/master/openfpga_flow/tasks/fpga_verilog/spypad/config/task.conf

Thank you for your answer. I changed `is_io=false` attribute and added a `is_data_io` since it's mandatory for iopad types. Now the circuit model looks like this:

```
<circuit_model type="iopad" name="axi_slave" prefix="axi_slave" verilog_netlist="./openfpga_cell_library/axi_slave_top.v">
            <design_technology type="cmos"/>
            <input_buffer exist="false" circuit_model_name="INVTX1"/>
            <output_buffer exist="false" circuit_model_name="INVTX1"/>
            <!-- input -->
            
            <port type="input" prefix="aw_addr"  lib_name="aw_addr"  size="32" is_global="true" is_io="true" is_data_io="true"/>
            <port type="input" prefix="aw_prot"  lib_name="aw_prot"  size="3"  is_global="true" is_io="true"/>
            <port type="input" prefix="aw_valid" lib_name="aw_valid" size="1"  is_global="true" is_io="true"/>
            <port type="input" prefix="w_data"   lib_name="w_data"   size="32" is_global="true" is_io="true"/>
            <port type="input" prefix="w_strb"   lib_name="w_strb"   size="4"  is_global="true" is_io="true"/>
            <port type="input" prefix="w_valid"  lib_name="w_valid"  size="1"  is_global="true" is_io="true"/>
            <port type="input" prefix="b_ready"  lib_name="b_ready"  size="1"  is_global="true" is_io="true"/>
            <port type="input" prefix="ar_addr"  lib_name="ar_addr"  size="32" is_global="true" is_io="true"/>
            <port type="input" prefix="ar_prot"  lib_name="ar_prot"  size="3"  is_global="true" is_io="true"/>
            <port type="input" prefix="ar_valid" lib_name="ar_valid" size="1"  is_global="true" is_io="true"/>
            <port type="input" prefix="r_ready"  lib_name="r_ready"  size="1"  is_global="true" is_io="true"/>
            <!-- output -->
            <port type="output" prefix="b_resp"   lib_name="b_resp"   size="2" is_global="true"   is_io="true"/>
            <port type="output" prefix="b_valid"  lib_name="b_valid"  size="1" is_global="true"   is_io="true"/>
            <port type="output" prefix="aw_ready" lib_name="aw_ready" size="1" is_global="true"   is_io="true"/>
            <port type="output" prefix="w_ready"  lib_name="w_ready"  size="1" is_global="true"   is_io="true"/>
            <port type="output" prefix="ar_ready" lib_name="ar_ready" size="1" is_global="true"   is_io="true"/>
            <port type="output" prefix="r_data"   lib_name="r_data"   size="32" is_global="true"  is_io="true"/>
            <port type="output" prefix="r_resp"   lib_name="r_resp"   size="2" is_global="true"   is_io="true"/>
            <port type="output" prefix="r_valid"  lib_name="r_valid"  size="1" is_global="true"   is_io="true"/>
            
            <port type="input" prefix="preg0_i"     lib_name="preg0_i"    size="32" />
            <port type="input" prefix="preg0_we_i"  lib_name="preg0_we_i" size="1" />
            <port type="input" prefix="preg1_i"     lib_name="preg1_i"    size="32" />
            <port type="input" prefix="preg1_we_i"  lib_name="preg1_we_i" size="1" />
            <port type="input" prefix="preg2_i"     lib_name="preg2_i"    size="32" />
            <port type="input" prefix="preg2_we_i"  lib_name="preg2_we_i" size="1" />
            <port type="input" prefix="preg3_i"     lib_name="preg3_i"    size="32" />
            <port type="input" prefix="preg3_we_i"  lib_name="preg3_we_i" size="1" />
            <port type="input" prefix="preg4_i"     lib_name="preg4_i"    size="32" />
            <port type="input" prefix="preg4_we_i"  lib_name="preg4_we_i" size="1" />
            <port type="input" prefix="preg5_i"     lib_name="preg5_i"    size="32" />
            <port type="input" prefix="preg5_we_i"  lib_name="preg5_we_i" size="1" />
            
            <port type="output" prefix="preg0_o" lib_name="preg0_o" size="32" />
            <port type="output" prefix="preg1_o" lib_name="preg1_o" size="32" />
            <port type="output" prefix="preg2_o" lib_name="preg2_o" size="32" />
            <port type="output" prefix="preg3_o" lib_name="preg3_o" size="32" />
            <port type="output" prefix="preg4_o" lib_name="preg4_o" size="32" />
            <port type="output" prefix="preg5_o" lib_name="preg5_o" size="32" />

            <port type="clock" prefix="clk_i" lib_name="clk_i" size="1"/>
            <port type="input" prefix="rst_ni" lib_name="rst_ni" size="1"/>
        </circuit_model>
```

But now, I get this error: 
`ERROR - -->>Error 1: Miss a default circuit model for the type iopad! Try to define it in your architecture file!`

I don't have an SRAM port, where should I add a circuit model?

tangxifan, 2024-07-07:
@yunuseryilmaz18 You may need to add the ``is_default`` attribute to the circuit model. See explanation here: https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#circuit-model

yunuseryilmaz18, 2024-07-18:
> @yunuseryilmaz18 You may need to add the `is_default` attribute to the circuit model. See explanation here: https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_library/#circuit-model

I added `is_default` attribute and changed the circuit model type to `hard_logic` too. Now, the fabric is generated without errors. Closing the issue now. Thank you @tangxifan.

closed 2024-07-18 by yunuseryilmaz18