lnis-uofu/OpenFPGA #1749: Does all parameters in task.conf file work?

chengquan, 2024-07-15:
Hi Xifan,

Many thanks to your excellent project.
May I know if all parameters in task.conf file work?
For instance, for the [General] part in .conf file, the parameter “verliog_output” seems to be unnecessary.
It is not used in the entire enviroment, right? Did I get it wrong?

In constrast, the parameter "openfpga_verilog_output_dir" is used in the  generated "*.openfpga" file to generate verilog file withe the command "write_fabric_verilog".
After reading your code, I am a little bit confused about part.
Could you plz explain it? 

Thanks,
Quan

tangxifan, 2024-07-15:
@chengquan Thank you for your interests. Indeed, there are redundant parameters in the task configuration file. during the aggressive upgrades in the user interface (when we switch from command-line options to the openfpga shell). But now, it is stable.

## Regarding the general settings 

- ``verilog_output`` does not matter. 
-  ``spice_output`` does not matter
- ``power_analysis`` matters. When enabled, ACE2 will be included in flow run. But not suggested when your have heterogeneous blocks in your architecture (due to ACE2's limitation for now).
- ``timeout_each_job`` is o.k. to adjust
- ``verific`` is still under test for now. You need a valid licence from verific and create your own Yosys version with verific plug-in.

## The recommended way

You can specify any parameters in your openfpga shell script ``.openfpga`` to customize your options. For example,

- You define a variable ``${OPENFPGA_VERILOG_OUTPUT_DIR}`` in your ``.openfpga`` file, you can assign a value in the task configuration file through ``openfpga_verilog_output_dir=<string>``.

As such, you can manipulate your flow in details.

chengquan, 2024-07-16:
Got it. I appreciate your help. have a nice day.

closed 2024-07-16 by chengquan