lnis-uofu/OpenFPGA #1752: ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH

PradyumnaG, 2024-07-20:
INFO (     MainThread) - Set up to run 2 Parallel threads
 INFO (     MainThread) - Currently running task basic_tests/generate_fabric
 INFO (     MainThread) - Created "run038" directory for current task run
 INFO (     MainThread) - Running "vpr_blif" flow
 INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
 INFO (     MainThread) - Created total 1 jobs
/home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:545: DeprecationWarning: currentThread() is deprecated, use current_thread() instead
  thread_name = threading.currentThread().getName()
/home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:545: DeprecationWarning: getName() is deprecated, get the name attribute instead
  thread_name = threading.currentThread().getName()
ERROR (00_and2_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_and2_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 573, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
subprocess.CalledProcessError: Command 'python3 /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga/src/k4_N4_90nm.xml /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.blif --top_module and2 --run_dir /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tasks/basic_tests/generate_fabric/run038/k4_N4_90nm/and2/MIN_ROUTE_CHAN_WIDTH --fpga_flow vpr_blif --openfpga_shell_template /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/openfpga_shell_scripts/generate_fabric_example_script.openfpga --openfpga_arch_file /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga/src/k4_n4_openfpga_arch.xml --openfpga_sim_setting_file /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --openfpga_verilog_output_dir /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tasks/basic_tests/generate_fabric/latest/k4_N4_90nm/and2/MIN_ROUTE_CHAN_WIDTH --activity_file /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --base_verilog /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v --power --power_tech /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --arch_variable_file /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tasks/basic_tests/generate_fabric/design_variables.yml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tasks/basic_tests/generate_fabric/run038/k4_N4_90nm/and2/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port --flow_config /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/tasks/basic_tests/generate_fabric/config/task.conf --default_tool_path /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/scripts/../misc/fpgaflow_default_tool_path.conf --TOP and2 --ACT /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.act --VERILOG /home/pradyumna/Documents/vtr_2/OpenFPGA/openfpga_flow/benchmarks/micro_benchmark/and2/and2.v' returned non-zero exit status 0.

I'm getting the above error while trying to run a task:python3 openfpga_flow/scripts/run_fpga_task.py basic_tests/generate_fabric

I'm sharing my vpr_stdout.log and openfpgashell.log files:

[vpr_stdout.log](https://github.com/user-attachments/files/16320734/vpr_stdout.log)
[openfpgashell.log](https://github.com/user-attachments/files/16320738/openfpgashell.log)

tangxifan, 2024-07-21:
@PradyumnaG I believe the root of cause is on the python version. If you are using a latest version, you may see the deprecation on some syntax in the run_fpga_flow.py. 

- You may use an older version. For example, python 3.10
- We will fix this issue soon and you do not need to worry about version requirements.

PradyumnaG, 2024-07-21:
I'm using Python 3.10.12. It works for the example arch files given(ones present in vpr_arch and openfpga_arch). I was trying to create these architecture files from the given documentation(K4N4 arch).

tangxifan, 2024-07-21:
@PradyumnaG If you have modified the architecture files for the testcase `generate_fabric`, the error should come from the architecture file. It complains about an invalid circuit model for your SRAM models. 

[image]

If you can share your arch file, it may help in debugging.

PradyumnaG, 2024-07-21:
Attaching the vpr arch and openfpga arch files
[k4_n4_openfpga_arch.txt](https://github.com/user-attachments/files/16325099/k4_n4_openfpga_arch.txt)
[k4_N4_90nm.txt](https://github.com/user-attachments/files/16325100/k4_N4_90nm.txt)
The simulation file is: auto_sim_openfpga.xml

tangxifan, 2024-07-22:
```
<circuit_model type="iopad" name="iopad" prefix="iopad" spice_netlist="io.sp" verilog_netlist="io.v">
  <design_technology type="cmos"/>
  <input_buffer exist="true" circuit_model_name="INVTX1"/>
  <output_buffer exist="true" circuit_model_name="INVTX1"/>
  <pass_gate_logic circuit_model_name="TGATE"/>
  <port type="inout" prefix="pad" size="1" is_global="true" is_io="true" is_data_io="true"/>
  **<port type="sram" prefix="en" size="1" mode_select="true" circuit_model_name="ccff" default_val="1"/>**
  <port type="input" prefix="outpad" size="1"/>
  <port type="output" prefix="inpad" size="1"/>
</circuit_model>
```

@PradyumnaG In your openfpga arch file, please be aware that you use an invalid circuit model ``ccff``, which is commented out.

PradyumnaG, 2024-07-22:
Thanks a lot, it works now. It was my bad I directly copied the circuit model from the documentation without seeing it.

closed 2024-07-22 by PradyumnaG