lnis-uofu/OpenFPGA #1884: iverilog Verification Task Fails with Module Not Found error

V-Geetha, 2024-11-04:
I am encountering an error while running the iverilog verification task for my adder design in the OpenFPGA flow. Although the relevant Verilog files are present in the specified SRC directory, I am receiving an error indicating that the root module adder_8_top_formal_verification_random_tb cannot be found.

**ERROR OUTPUT**

INFO (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - Launching iverilog_verification
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - iverilog_verification run failed with returncode 1
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - command iverilog -o compiled_adder_8 ./SRC/adder_8_include_netlists.v -s adder_8_top_formal_verification_random_tb -I ./SRC
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - -->>error: Unable to find the root module "adder_8_top_formal_verification_random_tb" in the Verilog source.
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - -->>1 error(s) during elaboration.
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - Current working directory : /fpga/OpenFPGA/openfpga_flow/tasks/fpga_verilog/adder/hard_adder/run021/k6_frac_N10_tileable_adder_chain_40nm/adder_8/MIN_ROUTE_CHAN_WIDTH
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - Failed to run iverilog_verification task
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - Exiting . . . . . .
ERROR (00_adder_8_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_adder_8_MIN_ROUTE_CHAN_WIDTH

Here is the list of files in the SRC directory where the error occurs
/fpga/OpenFPGA/openfpga_flow/tasks/fpga_verilog/adder/hard_adder/run021/k6_frac_N10_tileable_adder_chain_40nm/adder_8/MIN_ROUTE_CHAN_WIDTH/SRC
adder_8_autocheck_top_tb.v
adder_8_include_netlists.v
fabric_netlists.v
fpga_top.v
routing
adder_8_formal_random_top_tb.v
adder_8_top_formal_verification.v
fpga_defines.v
lb
sub_module

V-Geetha, 2024-11-08:
@tangxifan Thank you for your response and guidance!

 I've attached the adder_8_include_netlists.v file below.
 
 //-------------------------------------------
//      FPGA Synthesizable Verilog Netlist
//      Description: Netlist Summary
//      Author: Xifan TANG
//      Organization: University of Utah
//      Date: Tue Nov  5 13:00:49 2024
//-------------------------------------------
//----- Time scale -----
`timescale 1ns / 1ps

// ------ Include fabric top-level netlists -----
`include "./SRC/fabric_netlists.v"

`include "adder_8_output_verilog.v"

`include "./SRC/adder_8_top_formal_verification.v"
`include "./SRC/adder_8_formal_random_top_tb.v"
 
  Please let me know if there’s anything else I should check or if additional files would be helpful.

tangxifan, 2024-11-09:
@V-Geetha It looks like a mismatch in the top-level module name. Can you check if ``adder_8_top_formal_verification_random_tb`` is the module name defined in your ``/SRC/adder_8_formal_random_top_tb.v``?

V-Geetha, 2024-11-11:
//----- Default net type -----
`default_nettype none

module adder_8_top_formal_verification_random_tb;
// ----- Default clock port is added here since benchmark does >        reg [0:0] clk;

// ----- Shared inputs -------
        reg [0:0] a_0_;
        reg [0:0] a_1_;
        reg [0:0] a_2_;
        reg [0:0] a_3_;
        reg [0:0] a_4_;
        reg [0:0] a_5_;
        
        
        The top module name in the file /SRC/adder_8_formal_random_top_tb.v is indeed adder_8_top_formal_verification_random_tb. I’ve reviewed the the code and attested partial code above, and the top module name matches what’s declared in the file, so it’s correctly referenced. Is there any other files that has to be verified??

tangxifan, 2024-11-11:
@V-Geetha I think the Verilog netlists are clean. You may need to run iVerilog yourself and see what happened. It could be some syntax issue. Since you are not trying any testcase provided by OpenFPGA, it is difficult to debug from my side.

V-Geetha, 2024-11-18:
Thank you for your feedback . I will run iVerilog on my end to check for any syntax issues.

closed 2024-11-18 by V-Geetha