lnis-uofu/OpenFPGA #1900: Why is the multiple regions not applicable to ql_memory_bank-flatten protocol?

mustafaarslan0, 2024-11-15:
According to [OpenFPGA Configuration Protocol](https://openfpga.readthedocs.io/en/master/manual/arch_lang/config_protocol/#cmdoption-arg-num_regions), multiple configuration regions is not applicable to ql_memory_bank configuration protocol when BL/WL protocol flatten is selected. 

I have tried flatten protocol with different "num_regions" values, OpenFPGA did not generate any error and generated FPGA works well. Is there any bug here? Should I use different protocol if I need multiple regions? 

I am aiming big FPGA (like 32x32). I think I can not program all FPGA fabric, it is problematic.  So I need multiple regions to program a small piece of fabric at each program cycle. May you suggest physical design, power friendly way to do that?

Best regards.

tangxifan, 2024-11-15:
@mustafaarslan0 In the flatten protocol, all the BLs and WLs are exposed to you. It gives you the maximum freedom to manipulate the BLs and WLs. It is up to you to separate BLs and WLs into different regions.
Advanced settings for regions can be found in the [fabric key file](https://openfpga.readthedocs.io/en/master/manual/file_formats/fabric_key/#file-formats-fabric-key)

- If you can select to enable more than 1 WLs during each programming cycle.
- If you can split WLs in one row of BLs through fabric key, you can enable partial configuration during each programming cycle.

mustafaarslan0, 2024-11-18:
This is what I need. So there isn't any concern to use QL-MemoryBank-Flatten with multiple regions, documentation need to be updated.

Thanks.

closed 2024-11-18 by mustafaarslan0