lnis-uofu/OpenFPGA #1903: Reduce Unique Tiles in FPGA Fabric to Simplify Implementation Efforts

AhmadHouraniah, 2024-11-18:
**Is your feature request related to a problem? Please describe.**
The current implementation often leads to the generation of a large number of unique tiles based on the FPGA layout. However, when using a hierarchical approach, the area of each grid location is typically based on the largest tiles. This can result in significant additional implementation efforts with minimal gains.

**Describe the solution you'd like**
It would be highly beneficial to have an option to generate fewer unique tiles in the FPGA design. While this could involve adding extra hardware that isn't utilized in the final layout (e.g., unused routing logic in corner tiles), it would greatly reduce implementation efforts while maintaining the hierarchical approach.

For example, consider using a single CLB block throughout the FPGA fabric. In this case, tiles in the corners may contain routing logic that isn't used, but due to the grid layout, this would not impact the required die area. Ideally, we would want to produce only one unique tile for each type, such as CLBs, DSPs, BRAMs, and GPIOs.

The implementation of this feature would likely require support in both the RTL generation and the bitstream generation flows.
**Describe alternatives you've considered**
I've attempted to simplify the layout by making the routing logic more uniform, which reduced the number of unique tiles by about 50%. However, the FPGA design still includes approximately 18 unique tiles, which significantly increases the implementation workload. Each tile requires individual timing closure and implementation efforts, complicating the implementation process.

**Additional context**
Reducing the number of unique tiles would streamline the design flow, especially during implementation and timing closure. This change would make the FPGA design and implementation process more efficient and scalable, particularly for larger and more complex architectures.

tangxifan, 2024-11-21:
@AhmadHouraniah This is indeed a critical concern for many physical design engineers. OpenFPGA is continuously improving the tiling strategy. 
Here are the options that you may give a try

- Use adjust the number of pins of your CLB, DSP and BRAM. Try to have the same pin organization on each sides.
- Use the ``perimeter_cb`` feature in https://openfpga.readthedocs.io/en/master/manual/arch_lang/addon_vpr_syntax/#layout where you can even make your I/O tiles same. 

An ultimate way is to customize your routing resource graph.

AhmadHouraniah, 2024-11-25:
Thank you for your reply, I was able to reduce the number of tiles to 11 by considering the pin organizations, adjusting the different tiles to have the same number of output pins, and by adjusting the floorplan/layout of the FPGA.  

I would like to reduce this further if possible, how can I customize the routing resource graph? Are there any examples for this?

Thanks!