lnis-uofu/OpenFPGA #1958: Signal driving multiple modules causes bitstream error

narutozxp, 2025-03-03:
> **Describe the bug**
> A clear and concise description of what the bug is.

> **To Reproduce**
> Steps to reproduce the behavior:
> 1. Clone OpenFPGA repository and checkout commit id: <The problem commit id>
> 2. Execute OpenFPGA task or your own example: <more details>
> 3. See error

> **Expected behavior**
> A clear and concise description of what you expected to happen.

> **Screenshots**
> If applicable, add screenshots to help explain your problem.

> **Enviornment (please complete the following information):**

>  - OS: 

>  - Compiler:

> - Version:

> **Additional context**
> Add any other context about the problem here.

narutozxp, 2025-03-04:
I added a check statement at this location, and then the bitstream I generated seemed to work fine.
https://github.com/lnis-uofu/OpenFPGA/blob/68433d3a38d8a320bf1136bbdc78bb634642e82e/openfpga/src/annotation/vpr_routing_annotation.cpp#L60

The additional code added is as follows
```cpp
if(rr_graph.node_type(prev_node) == SINK) {
    return;
}
```

closed 2025-03-04 by narutozxp

tangxifan, 2025-03-04:
@narutozxp Thanks for the diagnose. I will patch.

tangxifan, 2025-03-04:
> I added a check statement at this location, and then the bitstream I generated seemed to work fine.
> 
> [OpenFPGA/openfpga/src/annotation/vpr_routing_annotation.cpp](https://github.com/lnis-uofu/OpenFPGA/blob/68433d3a38d8a320bf1136bbdc78bb634642e82e/openfpga/src/annotation/vpr_routing_annotation.cpp#L60)
> 
> Line 60 in [68433d3](/lnis-uofu/OpenFPGA/commit/68433d3a38d8a320bf1136bbdc78bb634642e82e)
> 
>  /* Ensure that the node_id is in the list */ 
> The additional code added is as follows
> 
> if(rr_graph.node_type(prev_node) == SINK) {
>     return;
> }

It is weird to see that a SINK node is not the ending point of an architecture. 
Do you customize the routing architecture?
Or did you find any existing architecture that you can reproduce the issue?

Thank you.
Xifan Tang

tangxifan, 2025-03-04:
I would like to include the benchmark you provide here as part of OpenFPGA. Do you have any license applied? If included, a MIT license will be applied.

tangxifan, 2025-03-12:
@narutozxp Great! Thank you for the contribution. I am impressed by your skills and work on OpenFPGA. Would you mind to share your email? Mine is xifan.tang@ieee.org 
There have been growing interests on OpenFPGA from industry. There could be many opportunities around. We can exchange idea if you like.