lnis-uofu/OpenFPGA #1968: Submodule vtr-verilog-to-routing does not default to openfpga branch

abdullahyildiz, 2025-03-16:
Hi,

I noticed that submodule [vtr-verilog-to-routing](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/8178b71295d2579c38403529765c95432c44bd0c) points to a different version in the master branch as opposed to what has been defined in [.gitmodules](https://github.com/lnis-uofu/OpenFPGA/blob/master/.gitmodules).

Is [openfpga](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/openfpga) branch of vtr-verilog-to-routing used for a different purpose?

tangxifan, 2025-03-17:
@abdullahyildiz The openfpga feature branch contains a number of experimental new features which VTR does not have. These  have enabled OpenFPGA.

We are working with VTR team to merge the PR: https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2135
where you can find a detailed description of new features.

abdullahyildiz, 2025-03-18:
Thank you for the clarification.

closed 2025-03-18 by abdullahyildiz