lnis-uofu/OpenFPGA #1994: Multiple local connections on same output in Switch Box

mustafaarslan0, 2025-05-09:
> **Describe the bug**
> There are multiple local connections on same signal in OpenFPGA generated Switch Box verilog netlist. My related architectural settings are:
```
<layout tileable="true" through_channel="true" opin2all_sides="false" concat_wire="true" concat_pass_wire="false">

<switch_block type="subset" fs="3" sub_type="wilton" sub_fs="3"/>
```
> Which part of OpenFPGA is buggy
[ ] Documentation
[ ] OpenFPGA flow
[X] FPGA-Verilog
[ ] FPGA-Bitstream
[ ] FPGA-SDC
[ ] FPGA-SPICE
[ ] VPR
[ ] Yosys

> **Screenshots**
> 

[image]

[image]

tangxifan, 2025-05-09:
@mustafaarslan0 This looks a critical bug to me. Can you provide more details? It is hard to tell where the bugs may come from. If you can share a simple testcase to reproduce the issue. I am happy to help.

mustafaarslan0, 2025-05-13:
Thanks for your attention. 

I use "subset" switch block with "wilton" sub_type. It has best area and low path delay-W combination according to your study "Switch Block Patterns for Tileable FPGA Routing Architectures"

To reproduce issue:
- Change this line with : <switch_block type="subset" fs="3" sub_type="wilton" sub_fs="3"/>
https://github.com/lnis-uofu/OpenFPGA/blob/cd5c82d45e2588775e8fc2e71a550d9cf79bc115/openfpga_flow/vpr_arch/k4_N4_tileable_40nm.xml#L139
- To run VPR with 4x4 layout and 300 routing segment, add options to openfpga script: --route_chan_width 300 --device 4x4
https://github.com/lnis-uofu/OpenFPGA/blob/cd5c82d45e2588775e8fc2e71a550d9cf79bc115/openfpga_flow/openfpga_shell_scripts/example_script.openfpga#L3
- Run task and check assignment statements in "sb_0__1_.v".

tangxifan, 2025-05-17:
@mustafaarslan0 Thanks for the details. Let me try the case.