lnis-uofu/OpenFPGA #2128: Modelsim run full testbench issue

zliu1Charlotte, 2025-09-04:
I am running RTL simulation 
I am able to run the simulation on both full and formal testbench without any issues using iverilog.
The full testbench loads the bitstream bit by bit and the formal testbench loads the bitstream by using $deposit or $force.

But when I run RTL simulation using Modelsim, it gives me the following error.
I know for sure when I generate gate-level netlist (after synthesis the RTL netlist), the full testbench worked well.

Could you confirm if the full testbench can be ran on the RTL netlist by loading the bitstream bit by bit? 
Because this conbimational loop does not exist to my understanding. 

`
Info: start testing
#############  Autofindloop Analysis  ###############
#############  Loop found at time 47725 ns ###############
#############       Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_3/mux_tree_basis_input2_mem1_12_out @ sub-iteration 0 at Value unknown (../rtl/SRC/sub_module/muxes.v:800)
#############       Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_tree_basis_input2_mem1_12_out @ sub-iteration 0 at Value unknown (../rtl/SRC/sub_module/muxes.v:800)
#############      Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_tree_basis_input2_mem1_13_out @ sub-iteration 0 at Value unknown (../rtl/SRC/sub_module/muxes.v:718)
#############       Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_3/mux_tree_basis_input2_mem1_13_out @ sub-iteration 0 at Value unknown (../rtl/SRC/sub_module/muxes.v:718)
#############  Active process: Atomic Verilog @ sub-iteration 1
#############     Source: ../rtl/SRC/sub_module/muxes.v:717
#############  Active process: Atomic Verilog @ sub-iteration 1
#############     Source: ../rtl/SRC/sub_module/muxes.v:717
#############   Active process: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_l4_in_0_/TGATE_1_/<anonymous> @ sub-iteration 1
#############    Source: ../rtl/SRC/sub_module/mux_primitives.v:80
#############     Driving net: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_l4_in_0_/TGATE_1_/out from St1 to St0
#############   Active process: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_3/mux_l4_in_0_/TGATE_1_/<anonymous> @ sub-iteration 1
#############    Source: ../rtl/SRC/sub_module/mux_primitives.v:80
#############     Driving net: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_3/mux_l4_in_0_/TGATE_1_/out from St1 to St0
#############      Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/INVTX1_14_/out @ sub-iteration 2 at Value St0 (../rtl/SRC/sub_module/inv_buf_passgate.v:38)
#############       Signal: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_3/INVTX1_14_/out @ sub-iteration 2 at Value St0 (../rtl/SRC/sub_module/inv_buf_passgate.v:38)
#############   Active process: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_l4_in_0_/TGATE_1_/<anonymous> @ sub-iteration 3
#############     Source: ../rtl/SRC/sub_module/mux_primitives.v:80
#############     Driving net: /maskmul_autocheck_top_tb/FPGA_DUT/grid_clb_1__1_/logical_tile_clb_mode_clb__0/mux_fle_1_in_2/mux_l4_in_0_/TGATE_1_/out from St1 to St0
################# END OF LOOP #################
############# ** Error (suppressible): (vsim-3601) Iteration limit 10000000 reached at time 47725 ns.
############# End time: 16:46:42 on Sep 04,2025, Elapsed time: 0:00:06
############# Errors: 1, Warnings: 0
`

zliu1Charlotte, 2025-09-04:
@tangxifan 
Thank you, 
could you provide examples on the 2 points? thanks!

tangxifan, 2025-09-04:
- https://openfpga.readthedocs.io/en/master/manual/arch_lang/circuit_model_examples/#multiplexer-with-constant-input
- https://openfpga.readthedocs.io/en/master/manual/openfpga_shell/openfpga_commands/fpga_verilog_commands/#cmdoption-include_timing